m-rtijn/mpu6050 #3: Calibration?

chrisspen, 2016-07-11:
How do you enter the offsets acquired during calibration? Otherwise, the values read from the device are raw and meaningless.

Tijndagamer, 2016-07-11:
Because no procedure for calibration of the hardware is given in the datasheet, there is no calibration implemented.

What does lack in this module is an implementation of the self test procedure (7.12 of the datasheet).

chrisspen, 2016-07-11:
I'm not asking how to calibrate it. This Arduino sketch explains how to do that http://wired.chillibasket.com/2015/01/calibrating-mpu6050/

I'm asking how to enter the 6 offsets (three for the accelerometer and three for the gyroscope). These are registers on the device:

MPU6050_RA_XA_OFFS_H
MPU6050_RA_YA_OFFS_H
MPU6050_RA_ZA_OFFS_H
MPU6050_RA_XG_OFFS_USRH
MPU6050_RA_YG_OFFS_USRH
MPU6050_RA_ZG_OFFS_USRH

If you don't enter these, the values read from the device will be useless.

Tijndagamer, 2016-07-11:
While developing this module I used the MPU-6050 Register Map and Descriptions file (Document number: RM-MPU-6000A-00, release date: 4.2, Release date: 2013-08-19) which does not include information about the offset registers. Because of this, there is currently no option to use these registers in this module, since I only _just_ learned about their existence (thanks to you because you pointed this out).

Tijndagamer, 2016-08-16:
I've decided that because for most applications calibrating the MPU-6050 is not necessary, since the factory calibration will be precise enough, that I won't add support for these calibration features.

If you want to add these features to this project, you're welcome to do so.

byapparov, 2021-01-09:
@m-rtijn I am not sure if I am doing something wrong, but I get bias of 16 d/s on the X gyro data. that makes it pretty much unusable without calibration.

Would be great to understand what is the way to overcome this, right now I have done a wrapper that adjusts for errors

Artoria2e5, 2023-04-27:
@byapparov  The Arduino-MPU6050 library (linked from the official site) does basically what you do: the `readNormalizeGyro()` function subtracts a fixed offset  and performs thresholding, both set from calibration at program start `calibrateGyro()`.

m-rtijn, 2023-04-27:
It seems that there also exists a Linux kernel module that implements interfacing with the MPU6050, that code can also be used as inspiration: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/iio/imu/inv_mpu6050?h=v6.3