m-rtijn/mpu6050 #5: Any possibility of a pypi release?

nickcoutsos, 2016-07-27:
I don't know how much interest you'd have in maintaining it, and I guess releases might not be so common anyway but it might be nice to provide this as a package on pypi for easy installation via pip.

I did it locally with the following _setup.py_:

```
from setuptools import setup

setup(
    name="MPU6050",
    version="1.0.0",
    url="https://github.com/Tijndagamer/MPU-6050-Python",
    py_modules=["MPU6050"],
    description=("Handles the I2C communication between a Raspberry Pi and a MPU-6050 Gyroscope/Accelerometer sensor."),
    author="MrTijn/Tijndagame",
    license="MIT"
)
```

I could be wrong but I think even just having that in the repo (no published package) would allow people to install via `pip install https://github.com/Tijndagamer/MPU-6050-Python` or maybe a git URL.

In any case, thanks very much for writing this module, I've just started playing around with the Pi and this has been a huge help.

Tijndagamer, 2016-07-28:
This is certainly a good idea and I'll look into it. Thanks for the feedback!

Tijndagamer, 2016-08-02:
[It is now packaged and available on pypi.python.org](https://pypi.python.org/pypi?name=mpu6050-raspberrypi&version=1.0&:action=display)

closed 2016-08-02 by Tijndagamer