m-rtijn/mpu6050 #14: Not defined

tmccoy33, 2017-03-20:
So I in fact am a little new to programming, however ive used sample code from here a few times successfully. I'm running into an issue with yours however. When I go to run the sample code with my mpu6050, it's telling me that line 6 "from mpu6050 import mpu6050" there is an import error and that no module exists. Its also said in your other code that mpu6050 line 2xx "mpu6050" is undefined. I am sure I'm missing something that I didn't install from you or something of this nature. Could you please provide some insight as to what could possibly fix my issue? Thank you.

Tijndagamer, 2017-03-20:
Could you copy the exact error?

Tijndagamer, 2017-03-20:
If you have a problem with the `smbus-cffi` module, remove [line 21 from setup.py](https://github.com/Tijndagamer/mpu6050/blob/master/setup.py#L21) and install the `python-smbus` package.

tmccoy33, 2017-03-20:
i removed line 1, it is saying this:

Traceback (most recent call last):
  File "/home/pi/mpu6050/bin/mpu6050-example.py", line 6, in <module>
    from mpu6050 import mpu6050
ImportError: No module named 'mpu6050'

tmccoy33, 2017-03-20:
sorry, i meant i removed line 21

Tijndagamer, 2017-03-21:
It seems like you didn't install it (correctly).

Run the following command in the root directory of the repository in order to install the module to your system:
```
sudo pyton setup.py install
```

closed 2017-04-01 by Tijndagamer

sutanu86, 2020-03-22:
same error..... 
 from mpu6050 import mpu6050
ImportError: No module named 'mpu6050'