m-rtijn/mpu6050 #15: I0Error

mspalluzzi, 2017-04-30:
Hey, seems like I've gotten the module installed correctly but when trying to run some sample code just to check if it will compile:

`from time import sleep
import RPi.GPIO as GPIO
from mpu6050 import mpu6050

sensor = mpu6050(0x68)

while True:
    accel_data = sensor.get_accel_data()
    gyro_data = sensor.get_gyrp_data()
    temp = sensor.get_temp()
`
I get this error:

[image]

Any ideas what that may be referring to / how to fix?
Looking into that directory it seems like all the files are properly placed.
Using Pi 3 if that changes anything.

Thanks!

Jimit04, 2017-04-30:
Run code with sudo.

Tijndagamer, 2017-04-30:
Thank you for using my module!

As @Jimit04 stated, you need to run the code using sudo to get access to the I2C bus.

closed 2017-04-30 by Tijndagamer

mspalluzzi, 2017-04-30:
Thanks for the quick response!
However, trying to run the code with sudo I'm getting the same error.

I did notice that I didnt do the pip install using sudo (seems like that would need elevated privileges?) so I went back through the whole process using sudo and I got the same error.

Tijndagamer, 2017-04-30:
Check if you have enabled I2C with raspi-config.

mspalluzzi, 2017-04-30:
There it is!

Thanks for the help and the great module!

Tijndagamer, 2017-04-30:
No problem.