m-rtijn/mpu6050 #17: IOE error

clark9501, 2018-02-26:
Dear Tijndagamer,
 when I set up the gyro, following error message popped up. 

  File "./gyro.py", line 6, in <module>
    sensor = mpu6050(0x68)
  File "/usr/local/lib/python2.7/dist-packages/mpu6050/mpu6050.py", line 60, in                                                                              __init__
    self.bus.write_byte_data(self.address, self.PWR_MGMT_1, 0x00)
  File "/usr/local/lib/python2.7/dist-packages/smbus/util.py", line 59, in valid                                                                             ator
    return fn(*args, **kwdefaults)
  File "/usr/local/lib/python2.7/dist-packages/smbus/smbus.py", line 145, in wri                                                                             te_byte_data
    raise IOError(ffi.errno)
IOError: 110

I made sure physical setting was correct, pre-req libraries are all installed and updated, I did a "i2cdetect -r 2" check and I can see 0x68. 

Can you please give any hint in terms of solving this problem?

Thanks!

Tijndagamer, 2018-02-26:
Hm. That's an error I haven't seen before I think. What hardware are you using (which Raspberry Pi)? Are there any mentions of an i2c error in the dmesg logs? Is the sensor powered correctly?

clark9501, 2018-02-28:
Thanks for the response! 

I am actually using beaglebone green as my platform. I think the sensor is powered correctly as we can see something fron the I2C test

Thanks.

Tijndagamer, 2018-02-28:
Alright, that makes things a little harder, since I have no experience with BeagleBone hardware myself. I think the I2C or GPIO might not be set up correctly by my mpu6050 library, since it's made for a Raspberry Pi. In order to fix this, I would suggest looking up how other people read data from an I2C sensor with a BBG, fork the library and make the appropriate changes.

Tijndagamer, 2018-03-10:
This other issue might be helpful for you: https://github.com/Tijndagamer/mpu6050/issues/15

closed 2018-03-10 by Tijndagamer