m-rtijn/mpu6050 #20: Python3 support

Tijndagamer, 2018-04-16:
Support python3 by switching to the smbus2 drop-in replacement for the python-smbus package.

The python3-dev branch should work with python3 and python2, but I can't test it at the moment.

bsimmo, 2018-04-25:
There should be no need to, smbus works fine under Python3.
There was a brief point a few years ago where it didn't.

The module is happily working on IDLE3 (from the same folder) on current Raspbian (Python3.5)
I've not checked for any maths errors, it didn't like the commenting the top though.

Tijndagamer, 2018-04-26:
> There should be no need to, smbus works fine under Python3.
There was a brief point a few years ago where it didn't.

As far as I know, the `python-smbus` package explicitly requires a python version <2.8; and thus doesn't support python3.

bsimmo, 2018-04-26:
We're not talking the package, but the smbus import.
Python3-smbus provides the Python3 version.

And I can test it and I have on recent Raspbian Stretch & Jessie.
So though I'd save you the old faff of using smbus2 as a drop in fix.

Bernix01, 2018-07-03:
any update on this?

Tijndagamer, 2018-07-03:
> any update on this?

There's a python3-dev branch which should work with smbus2, but I haven't been able to test it yet.

bsimmo, 2018-07-03:
It works with Python3.  What's the problem you are seeing?

Perhaps update your smbus (via apt python3-smbus or pip3 if that has it.).

Tijndagamer, 2018-07-03:
> It works with Python3. What's the problem you are seeing?

I haven't really had time to work on this project lately, but luckily the code generally seems to work so it's not really needed. However, last time I tried to use it I got some I/O errors, so I assume that there's something wrong with my sensor or with my Raspberry Pi (or the wires in between them).

Bernix01, 2018-07-03:
Just a syntax error found on the `python3-dev` `setup.py` on line 25 a missing comma at the end. With that fix it works. I don't think this needs a PR but if needed I'll do it.

Tijndagamer, 2018-07-04:
Thanks for the heads up about the missing comma @Bernix01, I fixed it in commit 20351ad045604919b318de7474b5e22bc36989c3.

Tijndagamer, 2018-07-29:
So I finally found some time to test it. Here are the results:
- The regular branch seems to work fine in python3 when the `python3-smbus` package is installed. I guess I had an outdated version of Raspbian the last time I worked on this project, because I don't remember ever having that package.
- The current `python3-dev` branch works with the smbus2 package. I'll rename the branch to `smbus2` for those who rather want to use smbus2.
So, as a conclusion: python2 and python3 are now supported. I'll make a new release (1.1) with an updated README shortly.

closed 2018-07-29 by Tijndagamer