m-rtijn/mpu6050 #22: 'module' object is not callable

Georg2K2, 2019-01-02:
when I try to run the example code:

from mpu6050 import mpu6050
sensor = mpu6050(0x68)
print(sensor.get_accel_data())

It prints out an error on line 2 stating: 'module' object is not callable
I've installed it correctly, and there's no error when importing the module

Tijndagamer, 2019-01-02:
Hi @Georg2K2, thank you for using my library!

I have only been able to reproduce this issue by incorrectly importing the module.

```python
>>> import mpu6050
>>> sensor = mpu6050(0x68)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'module' object is not callable
```

You can either import the module using `from mpu6050 import mpu6050` and call the class constructor with `mpu6050(0x68)` or import the module using `import mpu6050` and call the constructor with `mpu6050.mpu6050(0x68)`.

I hope this fixes your issue.

Georg2K2, 2019-01-03:
Like stated above, I import the module like:
from mpu6050 import mpu6050
and use it like described by you, but it still doesnt work for me :c

Tijndagamer, 2019-01-03:
Hm. That is really weird. Does the example script (in the repository `bin/mpu6050-example`) work for you?

Georg2K2, 2019-01-04:
I get the error: No module named 'mpu6050'
I tried running the setup.py again, sadly it doesn't work. (also you might edit the instructions from "python setup.py install" to "sudo python setup.py install")
I tried doing the pip install, restarted and now I get "TypeError: 'module' object is not callable" on the example script

Tijndagamer, 2019-01-04:
Which version of python are you using? If your system has both python 2.7.x and python 3, are you consistently using the same python version while testing?

Georg2K2, 2019-01-04:
im using python python 3

Georg2K2, 2019-01-04:
but I got it working now on Python 2.7.13!

Tijndagamer, 2019-01-05:
Nice that it works now, could you maybe explain what you did differently with python 2.7? In the case that someone else might ever stumble on the same issue as you.

closed 2019-01-05 by Tijndagamer

Georg2K2, 2019-01-05:
I just launched the script with python 2 instead of 3... ¯\_(ツ)_/¯

tmubashshira, 2021-04-04:
> I just launched the script with python 2 instead of 3... ¯_(ツ)_/¯

can you please tell the command you used to launch python 2 in raspberry pi??? I'm having the same error.....

tmubashshira, 2021-04-04:
> I just launched the script with python 2 instead of 3... ¯_(ツ)_/¯

if you're using python 2, doesn't it create problem in other operations???

GRhOGS, 2021-04-04:
I didn't run into any other problems... I haven't used that pi in years, I would have to google it again as well, sorry :/ maybe something like 'python2.7' in cmd, I'm not a huge Ubuntu buff...