m-rtijn/mpu6050 #23: no checking connectivity

Maxcullen, 2019-02-22:
after cloning the file, everything works but i don't get the line that says checking connectivity... done.
the next line 'python setup.py install' seems to have no action then because it cannot find the file

Tijndagamer, 2019-02-22:
What host device are you using? Could you share the output you're seeing?

Maxcullen, 2019-02-22:
I'm using a raspberry pi zero w. I am a bit of a noob so I apologize for my ignorance. I think it may not matter cause I went on with the programming and so far it looks good but I haven't had a chance to finish it all yet. I'd attach a photo but I can't get it up till tomorrow and didn't take one before. 
I really appreciate the assistance!

closed 2020-01-24 by m-rtijn