m-rtijn/mpu6050 #24: Gyro and accelerometer range

gempain, 2019-02-22:
Hello,

In [`mpu6050.py`](https://github.com/Tijndagamer/mpu6050/blob/master/mpu6050/mpu6050.py) you seem to always be reading raw values using the raw range in `get_gyro_data()` and `get_accel_data()`:
```python
 def read_accel_range(self, raw = False):
        raw_data = self.bus.read_byte_data(self.address, self.ACCEL_CONFIG)
        if raw is True:
            return raw_data
        elif raw is False:
            ...

def get_accel_data(self, g = False):
        accel_range = self.read_accel_range(True)
```
So I was wondering what is the point of having such an if/else code block ?

All the best;

Tijndagamer, 2019-02-22:
Even though the library itself never changes this `raw` option, I wanted to make it possible to easily get the raw values if one wants while using the library. Since raw is `False` by default, most users of the library can completely ignore this feature.

closed 2019-02-22 by Tijndagamer

gempain, 2019-02-22:
Sweet ! Thank you for the answer, and also thanks for the great plugin ! Really useful and well written.

Tijndagamer, 2019-02-22:
Thank you for the compliment :) Happy to hear that it is useful.