m-rtijn/mpu6050 #27: OSError: [Errno 121] Remote I/O error

kodecreer, 2019-09-19:
```
  File "/home/pi/.local/lib/python3.7/site-packages/mpu6050/mpu6050.py", line 60, in __init__
    self.bus.write_byte_data(self.address, self.PWR_MGMT_1, 0x00)
OSError: [Errno 121] Remote I/O error
```
I2C is enabled

mdsimmo, 2019-11-26:
For me, this error was because I had connected the mpu up to the wrong pins

toonijn, 2020-04-05:
The raspberry has two I²C busses. 

bus 0: pins 27, 28
bus 1: pins 3,5 (default)

If you connected the mpu6050 to 27 and 28 you will have to specify `bus=0`:
```
from mpu6050 import mpu6050
sensor = mpu6050(0x68, bus=0)
accelerometer_data = sensor.get_accel_data()
```

If you're using the wrong bus you'll also get the OP's error.

For more info about the pins see: https://pinout.xyz/pinout/i2c

m-rtijn, 2020-04-05:
Hey I didn't know this. Thanks @toonijn!

danchendrickson, 2021-11-10:
I am seeing the same error, but on the second time I run my code;  works fine on start up and first run.  The first time, the sensors works, and I can read data.  If I stop the code to make a change (been tweaking how it writes to file), when I restart it, the second time I get the error.  Works fine on start up and first run.

m-rtijn, 2021-11-19:
@danchendrickson I2C gets picky if you don't close everything neatly, I think that could cause your issue.

maryam0707, 2022-01-22:
bruh i am having the same issue of Errno 121 Remote I/O error for a while now can someone pls help me man im desperate asf

maryam0707, 2022-01-22:
I am getting the same error as Errno 121 Remote I/O error even after I have the correct addresses shown with the i2cdetect command.

spaceKelan, 2022-06-26:
I have trouble using your lib and respective sensort with other i2c devices. I think the problem is the smbus not being closed properly after each interaction on the bus. I am a newbie as well. I looked in another lib where the bus.close()  function is called. I wonder if you could have an example explaining this, because I get an I/O error every time. Thanks for the lib, it helped me already a lot!

kodecreer, 2022-08-18:
> The raspberry has two I²C busses.
>
> bus 0: pins 27, 28 bus 1: pins 3,5 (default)
>
> If you connected the mpu6050 to 27 and 28 you will have to specify `bus=0`:
>
> ```
> from mpu6050 import mpu6050
> sensor = [...]

Sorry for such a late reply, I was using pins 3 and 5, and it works with i2cdetect -y 1 pulling up address 0x68 as expected. 

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --    

Just right when I try to load it in  the library is when it has issues (even still after over 3 years) it still has the same problem).

kodecreer, 2022-08-20:
Alright so the answer was super simple. I thought it was because my soldering was bad or the wire length, but it wasn't the case. It was simply the fact that it's cheap garbage and turned out to be faulty.

 Since it's pretty cheap I just bought a new set and it worked perfectly fine.

closed 2022-08-20 by kodecreer