m-rtijn/mpu6050 #28: Raspberry Pi - No Accelerometer Data Detected

cherbin, 2020-01-10:
The raspberry pi is always getting zeros for the Accelerometer data.
Accelerometer data. Do you know the reason ?
x: 0.0
y: 0.0
z: 0.0
Gyroscope data
x: -156.183206107
y: 0.0
z: 0.0
Temp: 43.1182352941 C

root@raspberrypi:/opt/CANBUS/GYRO/py3/6050/mpu6050/bin#  sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

m-rtijn, 2020-01-12:
Hm. I've never had this problem before. Maybe your MPU-6050 is defective? Do you maybe have multiple to test if the problem persists?

m-rtijn, 2023-04-27:
I am closing this issue due to lack of response

closed 2023-04-27 by m-rtijn