m-rtijn/mpu6050 #29: ModuleNotFoundError when importing mpu6050

chancelier, 2020-03-06:
the smbus installation worked, I was able to test the i2c and that works. I also got no errors when installing through pip install. However as soon as I try to run a script in Python3, it says that:
Traceback (most recent call last):
  File "/home/pi/Documents/gyrompu.py", line 1, in <module>
    import mpu6050 as mpu6050
ModuleNotFoundError: No module named 'mpu6050'

chancelier, 2020-03-07:
I've read some threads about similar issues and it looks like it's all coming from using Python 3

chancelier, 2020-03-07:
seems like the issue is within Python because it is working when I'm using a script that's located in the same directory as the mpu6050.py

sutanu86, 2020-03-22:
same issue

m-rtijn, 2020-03-23:
@sutanu86 Did you install the package? If so, how?

@chancelier did you use the same python version for installing the package as for running it?

romybompart, 2020-03-23:
Hi, I just released a version that might help for different platform: Jetson nano, raspberry pi, Beaglebone, pyboard, ODROID and more. 

It is based on mr Tijn package. 

https://github.com/romybompart/py_imu_mpu6050
https://pypi.org/project/py-imu-mpu6050/

Now that is working in different platform I will add self_calibration functions and
Magnetometer support

MarCixn, 2020-12-18:
git clone https://github.com/Tijndagamer/mpu6050.git

cd mpu6050

sudo python setup.py install

EbGu3, 2021-10-01:
Hello, you can see what is the directory that save the files when you install the mpu6050, for example in my case add in ./.local/lib/python2.7/site-packages... if you run you can try to run with python2 <filename.py>

JohnAllen, 2024-10-17:
> git clone https://github.com/Tijndagamer/mpu6050.git
> 
> cd mpu6050
> 
> sudo python setup.py install

This didn't work for me.

"ModuleNotFoundError: No module named 'smbus'"