m-rtijn/mpu6050 #34: OS Error : [Errorno 121] Remote I/O error

Tutur26, 2020-12-21:
Hello,
I'm a student and I'm trying to implement a gimbal servoed on 1 axis with cheap components. To do that I bought a MPU6050 and a motoreductor. I have succeded to get angle on the X axis. However, when I'm trying to use these values to servo my motor, and when my motor starts to turn, the error 121 occured. 

`Traceback (most recent call last):
  File "/home/pi/Documents/TIPE/Programme de stabilisation.py", line 224, in <module>
    accel_data=mpu.get_accel_data()
  File "/usr/local/lib/python3.7/dist-packages/mpu6050_raspberrypi-1.1-py3.7.egg/mpu6050/mpu6050.py", line 139, in get_accel_data
    x = self.read_i2c_word(self.ACCEL_XOUT0)
  File "/usr/local/lib/python3.7/dist-packages/mpu6050_raspberrypi-1.1-py3.7.egg/mpu6050/mpu6050.py", line 72, in read_i2c_word
    low = self.bus.read_byte_data(self.address, register + 1)
OSError: [Errno 121] Remote I/O error`
 
I tried to reduce the sample rate at the exit from the MPU6050, and to reduce the lenght of the wires as I have seen on other platefroms, but it doesn't work.

Let me know if you have any solutions !

P.S : I have seen the previous thread on this error but it wasn't the same issue I think ( I have connected the SCL and the SDA on the GPIO 2 and 3 )

xichen1997, 2021-02-07:
I meet the same problem. I use PCA9865, however, when my servo moves, the reading process will be interrupted. I also get the same ERROR IO 121. Have you find the solution?

Tutur26, 2021-02-07:
> I meet the same problem. I use PCA9865, however, when my servo moves, the reading process will be interrupted. I also get the same ERROR IO 121. Have you find the solution?

I think it was due to electromagnetic disruptions, therefore, I have put a ferrite on the MPU6050 wires and on the motor wires and it works. I hope it can work for you too.

Tutur26, 2022-03-22:
> > > I meet the same problem. I use PCA9865, however, when my servo moves, the reading process will be interrupted. I also get the same ERROR IO 121. Have you find the solution?
> > 
> > I think it was due to electromagnetic disruptions, therefore, I have put a ferrite on the MPU6050 wires and on the motor wires and it works. I hope it can work for you too.
> 
> I meet the same problem. I use Raspberry Pi. I also get the same ERROR IO 121. Excessive line length can lead to electromagnetic disruptions？（only 1 meter）

I'm not a specialist but I think so. During my researches I saw that I2C protocol (communication protocol of the MPU6050) was not concieved for long wired connexions. And my problem occured even with wires of 10-15cm of length. Maybe you should try to reduce the length or use capacitors or ferrites to ensure a better communication. I didn't find another way to avoid this kind of data losses for the moment.

killer1kay, 2023-05-25:
doe this work with klipper and im also getting same error after i moved my printer
pi@e5plus:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@e5plus:~ $ python3 mpu_6050.py
Traceback (most recent call last):
  File "/home/pi/mpu_6050.py", line 3, in <module>
    mpu = mpu6050(0x68)
  File "/home/pi/.local/lib/python3.9/site-packages/mpu6050/mpu6050.py", line 72, in __init__
    self.bus.write_byte_data(self.address, self.PWR_MGMT_1, 0x00)
OSError: [Errno 121] Remote I/O error